rusefi/rusefi #2520: linking error on jenkins and Andrey's windows desktop

rusefillc, 2021-04-04:
build/obj/engine_configuration.o:engine_configuration.cpp:(.rdata$.refptr.engineConfiguration[.refptr.engineConfiguration]+0x0): undefined reference to `engineConfiguration'

```
Compiling tooth_logger.cpp
Compiling log_field.cpp
Linking build/rusefi_test
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: build/obj/engine_configuration.o:engine_configuration.cpp:(.rdata$.refptr.engineConfiguration[.refptr.engineConfiguration]+0x0): undefined reference to `engineConfiguration'
collect2: error: ld returned 1 exit status
make: *** [rules.mk:172: build/rusefi_test] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.
```

rusefillc, 2021-04-04:
https://github.com/rusefi/rusefi/pull/2508

rusefillc, 2021-04-04:
https://github.com/rusefi/rusefi/commit/1dffcf5729d5fcf9d31efc6b7baf98373662dc57

mck1117, 2021-04-04:
what compiler version do you have? can you `arm-none-eabi-gcc -v`?

rusefillc, 2021-04-04:
that's unit test linking error

[image]

[image]

matches 9.2.0 in issue description

mck1117, 2021-04-04:
I would try to update if you can, I have `arm-none-eabi-gcc` 9.3.1, and `x86_64-linux-gnu-gcc` 10.2.0

rusefillc, 2021-04-04:
gcc bump to 10.2 made no difference

make after clean:
```
Compiling tooth_logger.cpp
Compiling log_field.cpp
Linking build/rusefi_test
/usr/lib/gcc/x86_64-w64-mingw32/10/../../../../x86_64-w64-mingw32/bin/ld: build/obj/engine_configuration.o:engine_configuration.cpp:(.rdata$.refptr.engineConfiguration[.refptr.engineConfiguration]+0x0): undefined reference to `engineConfiguration'
collect2: error: ld returned 1 exit status
make: *** [rules.mk:172: build/rusefi_test] Error 1
"make -j8 all" terminated with exit code 2. Build might be incomplete.

```

[image]

[image]

rusefillc, 2021-04-04:
is that second time we are hitting linker optimization being different between win and unix?

commit 2b094849 (2021-04-04): linking error on jenkins and Andrey's windows desktop fix #2520
closed 2021-04-04 by rusefillc