rusefi/rusefi #2526: So what do we want by default, pull-up or pull-down?

rusefillc, 2021-04-04:
It looks like comment says pulldown while value says PIN_PUPDR_PULLUP? Is there a bug or an error here?
```
/*
 * Default to input mode, with internal pulldown resistor enabled.
 */
#define EFI_DR_DEFAULT PIN_PUPDR_PULLUP
```

See also https://github.com/rusefi/rusefi/pull/2525

rusefillc, 2021-04-04:
@mck1117 @dron0gus @andreika-git what's your opinion?

dron0gus, 2021-04-04:
My opinion: each pin has its own safe state and the is no common solution for all pins (on all boards)
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/subaru_eg33/board_io.h

mck1117, 2021-04-04:
> the is no common solution for all pins (on all boards)

So what about firmware update? We have zero control when the DFU bootloader is running, so I think all pins have to be safe when floating (or weak pullup/down).

dron0gus, 2021-04-04:
>  so I think all pins have to be safe when floating (or weak pullup/down).

If HW is designed keeping this in mind - yes. Then we do not need to touch any gpio settings in early code. We can just wait for high level code to init pins...

mck1117, 2021-04-04:
> If HW is designed keeping this in mind - yes
afaik, proteus/MRE/hellen do this