rusefi/rusefi #2533: WOW major scheduling drama on NB2 running with Proteus

rusefillc, 2021-04-08:
While collecting reference "good" data using Proteus I've realized that we have major ignition imbalance?! Coil No 1 is fired less often than Coil No 2?

[image]

See https://rusefi.com/forum/viewtopic.php?p=40317#p40317 for more images

rusefillc, 2021-04-08:
#1563 would be really useful here

Writing logicdata on SD card would be even better

rusefillc, 2021-04-08:
@stefanst FYI

@CrazyStriker has 36 trigger on his NB2 and this has to be trigger related

rusefillc, 2021-04-15:
self stim at the moment
```
2021-04-15_17_33_28_384: EngineState: TRG  rpm=1200 time=2694 index=0: gap=0.799 expected from 0.349 to 0.980 error=No
2021-04-15_17_33_28_384: EngineState: TRG duty 0.000000000 0
2021-04-15_17_33_28_384: EngineState: TRG  rpm=1200 time=2694 index=0: gap=1.250 expected from 0.349 to 0.980 error=No
2021-04-15_17_33_28_487: EngineState: TRG duty 0.000000000 0
```

JimStim
```
2021-04-15_17_33_57_104: EngineState: TRG  rpm=763 time=2723 index=0: gap=0.636 expected from 0.349 to 0.980 error=No
2021-04-15_17_33_57_104: EngineState: TRG duty 0.000000000 0
2021-04-15_17_33_57_104: EngineState: TRG  rpm=763 time=2723 index=0: gap=1.571 expected from 0.349 to 0.980 error=No
2021-04-15_17_33_57_104: EngineState: TRG duty 0.000000000 0
```

trigger shape is wrong

rusefillc, 2021-04-15:
NB2 shape as is in rusEFI

[image]

rusefillc, 2021-04-15:
1200 jim

[image]

rusefillc, 2021-04-15:
[image]
[image]

rusefillc, 2021-04-15:
Self stim

[image]

mck1117, 2021-04-15:
I suspect we took the timing while cranking, which was wrong because the engine speed varies

rusefillc, 2021-04-15:
So rusEFI wiki was showing correct NB1 trigger wheel

but NB2 was not consistent with NB1 WTF. NB2 shape is now more correct

commit 5d09524f (2021-04-15): NB2 trigger decoder fix #2533
closed 2021-04-15 by rusefillc