rusefi/rusefi #2541: Specific board just does not save configuration / always reset to default on boot

rusefillc, 2021-04-12:
I have a microRusEFI which just does not save config changes

```
2021-04-11_22_37_30_561: EngineState: Writing pending configuration...
2021-04-11_22_37_30_562: EngineState: FLASH_SUCCESS
2021-04-11_22_37_30_562: EngineState: confirmation_writeconfig:11
```
but on next boot everything is back to defaults

rusefillc, 2021-04-15:
@mck1117 says

[image]

mck1117, 2021-04-15:
either this MCU is:
- fake
- failed QC yet made it out anyway (?)
- damaged from soldering (unlikely that it's damaged exactly in to being an F429VET)

rusefillc, 2021-04-15:
got this MCU from JLC

Thank you @mck1117 for the analysis I guess case closed

closed 2021-04-15 by rusefillc

mck1117, 2021-04-15:
I posted on eevblog for fun: https://www.eevblog.com/forum/microcontrollers/did-i-actually-find-a-defective-stm32/msg3549397/

rusefillc, 2021-04-15:
same or similar batch @
[image]

same or similar batch