rusefi/rusefi #2551: DRAM SDRAM

rusefillc, 2021-04-15:
Sweet RAM...
there are definitely legit use-cases for RAM

rusefillc, 2021-04-15:
https://docs.rs-online.com/b72f/0900766b815247be.pdf tells us that Discovery kit with STM32F429ZI MCU has used IS42S16400J chip

https://jlcpcb.com/parts/componentSearch?searchTxt=sdram

[image]

rusefillc, 2021-04-15:
See also https://github.com/andreika-git/hellen-one/issues/29

rusefillc, 2021-06-27:
That's community see SDRAMConfig

rusefillc, 2021-11-10:
I have pushed some not-yet-working code a while back and no time/brain to look into this further.

Just FYU @racer-coder STM32F429I-DISC1 is in stock at digikey and it has SDRAM on board, same board @mck1117 has.

racer-coder, 2021-11-10:
Why do we need more RAM?  It won't help older hardware, the F429 has 256k, some F7 have 512, and H743 has 1M.  Now that it's looking like even the 192k is plenty, what's the driver for this?

Downside - more variants to program, more boardspace, more complexity.

rusefillc, 2021-11-10:
No specific use-case, very long term plan.

But I can think of build-in bare bones oscilloscope and buffer transfer, crazy Lua, whatever self-tuning stuff with https://github.com/dpilger26/NumCpp and all other kinds of crackpipe dreams.

rusefillc, 2022-01-11:
Thank you @dron0gus !

https://rusefi.com/forum/viewtopic.php?p=43998#p43998

closed 2022-01-11 by rusefillc