rusefi/rusefi #2552: MCU says it's too cold

rusefillc, 2021-04-15:
hellen72 here running pretty fresh firmware

2021-04-14_20_40_49_996: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_50_216: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_50_434: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_50_656: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_50_768: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_50_983: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_51_203: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72
2021-04-14_20_40_51_422: EngineState: CRITICAL error: Invalid CPU temperature measured -54.900878906 20210407@hellen72

mck1117, 2021-04-15:
what PN is this MCU?

rusefillc, 2021-04-15:
@mck1117 it's F429

rusefillc, 2021-04-15:
I believe it was a random one time thing

same board was happy before on same pretty fresh firmware

I've updated to latest and got NB2 to run

mck1117, 2021-04-15:
I think it was briefly broken maybe a month ago, all the F42x I have report temp correctly.

rusefillc, 2021-04-21:
I am now experiencing same on real car definitely on latest firmware

battery might be not totally charged?

i am cranking for a bit and then

[image]

rusefillc, 2021-05-01:
We have user report with same error on Frankenso https://rusefi.com/forum/viewtopic.php?p=40719#p40719

rusefillc, 2021-06-21:
HW CI says Hi
```
 [junit] Testcase: testMcuTemperature took 1,562 sec
    [junit]     FAILED
    [junit]
    [junit] junit.framework.AssertionFailedError:
    [junit]     at com.rusefi.common.MiscTest.testMcuTemperature(MiscTest.java:27)

rusefillc, 2021-07-02:
```
    [junit] Testcase: testMcuTemperature took 0,015 sec
    [junit]     FAILED
    [junit] mcuTemp is NaN
    [junit] junit.framework.AssertionFailedError: mcuTemp is NaN
    [junit]     at com.rusefi.common.MiscTest.testMcuTemperature(MiscTest.java:29)
    [junit]
```

rusefillc, 2021-07-26:
@mck1117 Build failed in Jenkins: integration_and_primary_bundle #10129

rusefillc, 2021-08-04:
@mck1117 it happened again :(

```
    [junit] I 210803 221841.248 [Commands Queue] BinaryProtocol - Sending [enable test_mode]
    [junit] I 210803 221841.248 [main] IoUtil - Command [enable test_mode] executed in 0
    [junit] I 210803 221841.248 [Commands Queue] IoUtil - Got confirmation in 0ms
    [junit] ------------- ---------------- ---------------
    [junit]
    [junit] Testcase: testMcuTemperature took 8,719 sec
    [junit]     FAILED
    [junit] mcuTemp is NaN
    [junit] junit.framework.AssertionFailedError: mcuTemp is NaN
    [junit]     at com.rusefi.common.MiscTest.testMcuTemperature(MiscTest.java:29)
    [junit]
    [junit] Testcase: burn took 0,281 sec

BUILD FAILED
<http://127.0.0.1:8080/job/integration_and_primary_bundle/ws/java_console/build.xml>:196: Test com.rusefi.common.MiscTest failed

Total time: 16 seconds
I 210803 221841.357 [communication executor1] MessagesCentral - postMessage EngineState: [trigger_hw_input] disabled
I 210803 221841.357 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_disable trigger_hw_input:24
I 210803 221841.357 [communication executor1] MessagesCentral - postMessage EngineState: [test_mode] enabled
```

mck1117, 2021-08-04:
The ECU sends temperature to the PC as a scaled integer.  How does it end up NAN? Integers can't do NAN.

mck1117, 2021-08-04:
Here's your NAN:

[image]

rusefillc, 2021-08-08:
Thank you @mck1117 added some wait loop

mck1117, 2021-08-08:
why not just wait 1 second before quering temperature?

mi-hol, 2023-01-26:
the last comments sound like issue is solved, right?

mck1117, 2023-01-26:
we haven't seen it in forever

closed 2023-01-26 by mck1117