rusefi/rusefi #2553: Too many logs are zero size

rusefillc, 2021-04-15:
hellen72c

I would expect a higher percentage of logs to be not empty

[image]

mck1117, 2021-04-15:
We flush every 100 writes, and each write is 512 bytes, so files always come out as a multiple of 50k bytes.  At the default logging rate of 20hz, with each record around 100 bytes, that's 2k per second, so if you run for less than 25 seconds, the file will be zero length.

mck1117, 2021-04-15:
Whether that's the right behavior is up for debate, but that's why those files are all integer multiples of 50k bytes.

rusefillc, 2021-04-15:
I think we should flush more often in the beginning or until certain RPM is reached?

closed 2021-04-15 by rusefillc

OrchardPerformance, 2021-04-19:
Just a note on this one - We used the SD logging all weekend with no issues. Zero size logs were helpful to detect just a quick switch on to purge NOS or check something.
Any real runs were perfectly OK. 
Only time a Zero log could be an issue is if doing a cranking test but I would assume anyone doing that would have the laptop logging instead.

PR #2558 (2021-04-15): flush SD log 10x faster
fix #2553