rusefi/rusefi #2555: Catch up with Chibios 20.3

rusefillc, 2021-04-15:
Chibios has some H7 bugfixes it's time for us to catch up

rusefillc, 2021-04-15:
I've refreshed https://github.com/rusefi/ChibiOS/tree/stable_20.3.x

next step rebase https://github.com/rusefi/ChibiOS/tree/stable_20.3.x.rusefi on https://github.com/rusefi/ChibiOS/tree/stable_20.3.x

rusefillc, 2021-04-15:
@dron0gus can I interest you in this one?

mck1117, 2021-04-15:
Are there actually h7 changes? Afaik the only H7-specific change recently would have been the CAN driver changes that I wrote and already cherry picked in to our repo.

rusefillc, 2021-04-15:
http://forum.chibios.org/viewtopic.php?f=7&t=5430#p39428 does mention h7 and I assume it's just a healthy thing to catch up once in a while?

mck1117, 2021-04-15:
Fair enough.  Though it seems the FDCAN changes actually haven't been merged yet.

rusefillc, 2022-01-16:
https://forum.chibios.org/viewtopic.php?f=7&t=5430&p=40631#p40631 mentions H7 a couple more times

@mck1117 @dron0gus poke :)

rusefillc, 2022-01-16:
See also #3794

rusefillc, 2022-01-16:
Just refreshed https://github.com/rusefi/ChibiOS/tree/stable_20.3.x to match upstream

rusefillc, 2022-01-16:
just merged https://github.com/rusefi/ChibiOS/tree/stable_20.3.x into https://github.com/rusefi/ChibiOS/tree/stable_20.3.x.rusefi there were five conflicts but neither seemed to be about actual RE changes

rusefillc, 2022-01-16:
looks like a success!

closed 2022-01-16 by rusefillc