rusefi/rusefi #2564: Smarter pinout re-generation

rusefillc, 2021-04-18:
At the moment connector pinouts are generated nightly

Would be nice to have more quick response if we change activity, do we have any simple options?

chuckwagoncomputing, 2021-04-18:
The simplest option would be to run it on changes matching the path "firmware/config/boards/*"
That will run a lot when it doesn't need to. It would be perfect if we could do "firmware/config/boards/*/connectors/*", but I'm guessing that doesn't work.

rusefillc, 2021-04-18:
Can we have a button _somewhere_?

Can we simply write down the URL for GHA so that I can go there and click "rerun"? Even manual button would be a way for now.

chuckwagoncomputing, 2021-04-18:
I don't think there is a way to manually trigger a workflow run. You can re-run a workflow that has already run, but it will run on the same commit, not the newest.

chuckwagoncomputing, 2021-04-27:
Well now that I decided to not be too lazy to google, we have #2588

chuckwagoncomputing, 2021-04-27:
Do you see the button to run the workflow now?
Here are the instructions:
https://docs.github.com/en/actions/managing-workflow-runs/manually-running-a-workflow#running-a-workflow-on-github

rusefillc, 2021-04-27:
[image]

rusefillc, 2021-04-27:
That's my chance to advertise IDEA

Note syntax highlighting difference between valid and invalid

[image]

[image]

rusefillc, 2021-04-27:
https://github.com/rusefi/rusefi/actions/workflows/gen-pinouts.yaml

shows a button now!

[image]

rusefillc, 2021-04-27:
it's green!

No idea if anything was uploaded logging is a unclear?

[image]

rusefillc, 2021-04-28:
it works thank you!!!

closed 2021-04-28 by rusefillc