rusefi/rusefi #2565: hellen64 scripts and pinout

rusefillc, 2021-04-18:
We need scripts to generate hellen64 binaries https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen/hellen64_miataNA6_94

We also need to finish connector https://rusefi.com/docs/pinouts/hellen/hellen64_miataNA6_94/

rusefillc, 2021-04-18:
https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen/hellen72/connectors is a sample

BeerMoneyMotorsports, 2021-04-26:
Hellen64 pinout in progress.  Working on having it ready this week so we can get going with the H64 as soon as they arrive.

rusefillc, 2021-04-30:
@BeerMoneyMotorsports we need all the "id: GPIOxx" to be filled in like in h72 See https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen72/connectors/main.yaml

Can you trace Injector #1 in hellen72 into something called "PG7" probably? PG7 is stm32 pin name, and GPIOG_7 is how we represent that stm32 pin name in the yaml file

closed 2021-05-13 by rusefillc