rusefi/rusefi #2583: Only 24 lines are efiPrinted if do a lot of prints

dron0gus, 2021-04-25:
```
	for (int i = 0; i < 50; i++) {
		efiPrintf(" %d this is just a test!", i);
	}
```
Output:
```
2021-04-25_20_52_53_085: EngineState:  0 this is just a test!
2021-04-25_20_52_53_086: EngineState:  1 this is just a test!
2021-04-25_20_52_53_087: EngineState:  2 this is just a test!
2021-04-25_20_52_53_088: EngineState:  3 this is just a test!
2021-04-25_20_52_53_089: EngineState:  4 this is just a test!
2021-04-25_20_52_53_091: EngineState:  5 this is just a test!
2021-04-25_20_52_53_091: EngineState:  6 this is just a test!
2021-04-25_20_52_53_091: EngineState:  7 this is just a test!
2021-04-25_20_52_53_092: EngineState:  8 this is just a test!
2021-04-25_20_52_53_092: EngineState:  9 this is just a test!
2021-04-25_20_52_53_093: EngineState:  10 this is just a test!
2021-04-25_20_52_53_093: EngineState:  11 this is just a test!
2021-04-25_20_52_53_094: EngineState:  12 this is just a test!
2021-04-25_20_52_53_094: EngineState:  13 this is just a test!
2021-04-25_20_52_53_094: EngineState:  14 this is just a test!
2021-04-25_20_52_53_095: EngineState:  15 this is just a test!
2021-04-25_20_52_53_095: EngineState:  16 this is just a test!
2021-04-25_20_52_53_097: EngineState:  17 this is just a test!
2021-04-25_20_52_53_097: EngineState:  18 this is just a test!
2021-04-25_20_52_53_097: EngineState:  19 this is just a test!
2021-04-25_20_52_53_098: EngineState:  20 this is just a test!
2021-04-25_20_52_53_098: EngineState:  21 this is just a test!
2021-04-25_20_52_53_099: EngineState:  22 this is just a test!
2021-04-25_20_52_53_099: EngineState:  23 this is just a test!
```

This affects 'pins' command output. (and may be some other)

rusefillc, 2021-04-25:
@mck1117 shall we bump
constexpr size_t lineBufferCount = 24;

to 200?

rusefillc, 2021-04-27:
lets try 'pins' in master

dron0gus, 2021-05-05:
Looks like fixed.

PR #2592 (2021-04-27): higher priority for log buffer flush
fixes #2583
closed 2021-05-05 by rusefillc