rusefi/rusefi #2584: No Heater + on Hellen 72

BeerMoneyMotorsports, 2021-04-26:
no H+ marking on the Hellen72 board.  See attached pic.  possible misprint? 
[image]

rusefillc, 2021-04-26:
I wonder if H+ is supposed not to touch H72 due to the way OEM harness is done?

@andreika-git we need this documented on hellen72 public wiki please

BeerMoneyMotorsports, 2021-04-26:
are we meant to use oem wiring then?  current discussions have pointed to it being a new harness from the Hellen to the sensor.  not involving the oem wiring?  Im not seeing that detail in the schematic for H72.

andreika-git, 2021-04-26:
[image]

BeerMoneyMotorsports, 2021-04-26:
@andreika-git thank you for that illustration and the conversation last night to explain it.

closed 2021-04-26 by BeerMoneyMotorsports