rusefi/rusefi #2586: Change way we configure analog inputs

rusefillc, 2021-04-26:
it looks like at the moment we have too much unneeded complexity with too much of limitations in the analog input area

problem statement 1: it's unclear if 'adc_channel_e' concept is serving any purpose

problem statement 2: adc_channel_mode_e is currently hard-coded, open question if that should be the case or not

problem statement 3: it's really unclear how would we implement external ADC chips. Somewhat similar to external GPIO drama

rusefillc, 2021-04-26:
@dron0gus @mck1117 @andreika-git FYI

dron0gus, 2021-04-26:
What problem we are going to solve?
1. this is good representation for user.
2. this is good enought for board with fixed hardware configuration.
3. why not? actually we should not make any difference between on-chip ADC and external ADC. Just a ADC occupying given range of inputs and providing proper API.

I would rather think about more extended service for adc users. Like start/stop measurement and then giveAveragedValue, giveMin, giveMax, so on. Simple filters, maybe. Now few ADC users implement such stuff. Why not to move to common code.

rusefillc, 2021-04-26:
@dron0gus I am not aware of any .ini file using adc_channel_e to represent values to user. It's actually the opposite - we have a mechanism to hide adc_channel_e from the users.

So killing adc_channel_e would at least remove the translation-forward and translation-backwards code

dron0gus, 2021-04-26:
@rusefillc ok, for "advanced" user who ever opened sources or used rusefi console.

Killing adc_channel_e would cause a loooot of magic numbers in engine/boards configs. Magic numbers during debug and so on.

While I'm agree that adc_channel_e is not good for code that uses adc. Better to have pointer and skip translation. This will speedup code but increase memory footprint.

mck1117, 2021-04-27:
> problem statement 1: it's unclear if 'adc_channel_e' concept is serving any purpose

Please expand on this.  What purpose does adc_channel_e not serve?

rusefillc, 2021-04-27:
@mck1117 it's unclear if 'adc_channel_e' concept is serving any purpose - i.e. it looks like we can simply drop adc_channel_e, work with pin index directly and loose nothing. it looks like without adc_channel_e we would simply have less code and same exact result as we do today.

mck1117, 2021-04-27:
pin index? index of what? IMO the enum is valuable because it lets you indicate `NONE` without a magic int value

rusefillc, 2021-04-27:
@mck1117 edited "pin index" is now "brain_pin_e"
brain_pin_e has NONE
I am not suggesting
s/adc_channel_e/int/

I am suggesting s/adc_channel_e/brain_pin_e/