rusefi/rusefi #2594: Throw a warning if a redundant sensor is clamped

shadowm60, 2021-04-27:
In case one sensor fails, and the delta between these two sensors are > 5% we will return 0 as a failsafe, this should also post a warning in the log file to speed up the debugging of the issue (due to wiring or bad sensor).

rusefillc, 2021-04-27:
@shadowm60 there is a good chance you have looked at the code, can you please paste relevant code or file name here?

are you planning to PR or asking for help?

shadowm60, 2021-04-28:
Get method implementation of the Redundant class i think is the best place to set warning:
SensorResult RedundantFordTps::get() const 
SensorResult RedundantSensor::get() const

In these method whenever unexpected is return it is an error case.
@mck1117 what do you think?

shadowm60, 2021-04-28:
@rusefillc i can make a PR, i just need a confirmation from @mck1117 and maybe help on how to identify what kind of redundant sensor it is since we could throw:
OBD_Throttle_Position_Sensor_Range_Performance_Problem and P2109 Throttle/Pedal Pos SensorA Minimum Stop Perf ?

nmschulte, 2022-12-31:
@mck1117 did this ever get resolved; should we cover it with #4848?  I have some proposed changes almost ready to share, so can account for it if it's not and you can give the pointers re: _how?_.

mck1117, 2022-12-31:
yeah, we log a warning for `OBD_TPS1_Correlation`, which is 2135.