rusefi/rusefi #2597: too many users are getting confused by injector compensation mode

mck1117, 2021-04-28:
Lots of people should be setting "none", but are setting "fixed rail pressure"

blundar, 2021-05-14:
This seems like a descriptive problem?

No compensation = return style regulator
Fixed rail pressure = mechanical returnless fixed pressure
rail pressure sensor, MAP = physical sensor measuring rail pressure, physical sensor measuring manifold pressure
differential sensor = physical sensor connected to both rail and manifold (Ford/Bosch style) measuring differential pressure directly

mck1117, 2021-05-14:
We should put that text in the tool tip

rusefillc, 2022-01-04:
@BeerMoneyMotorsports @blundar @OrchardPerformance this came up again. If we do not work smart we would have to work hard.

[image]

BeerMoneyMotorsports, 2022-01-04:
This was my mistake as i failed to read the tool tip.  @OrchardPerformance has already done his part and this ticket should be closed.  

[image]

closed 2022-01-04 by rusefillc