rusefi/rusefi #2627: Break out unit tests

rusefillc, 2021-05-06:
in order to avoid dead slow linking would it be beneficial to split unit tests into a bunch of separate Makefile projects?

mck1117, 2021-05-06:
```
rusefi/unit_tests$ rm -rf .dep build
rusefi/unit_tests$ time make -j20
....
real    1m52.901s
user    6m27.688s
sys     7m25.625s
```
then when touching one file (the "dev loop"):
```
rusefi/unit_tests$ time make -j20
Compiling main.cpp
Linking build/rusefi_test

real    0m17.057s
user    0m8.141s
sys     0m8.359s
```

rusefillc, 2021-05-06:
my OS might be holding me back :(

mck1117, 2021-05-06:
This is using WSL (I think it's even WSL v1), btw. It's significantly faster still on my slower-hardware-but-actual-Linux machine.

mck1117, 2021-05-06:
real linux machine says
clean build: 1m5s
touch main.cpp: 6.7s

mck1117, 2021-05-06:
setting `USE_LINK_GC = no` on WSL machine says
clean: 1m39s
main.cpp: 12.3s

rusefillc, 2021-05-07:
[image]

rusefillc, 2021-05-07:
yes master

mck1117, 2021-05-07:
so problem solved?

rusefillc, 2021-05-07:
problem is not solved

1) improving rusEFI design is still a valuable goal and I would like to spend a bit more time on the POC of isolated compilation. Isolated compilation could be used by CI to assert design quality

2) I still do not have a usable overall setup on WSL just command line compilation at the moment

2.1) I still have not memories if it's WSL or WLS :(

mck1117, 2021-05-07:
1. so perhaps a better issue would be "tear down `Engine` god object"
2. you can `wsl bash -l path_to_script.sh` from anywhere in windows and it'll do the right thing, for easy integration with IDE's build button.

mck1117, 2021-10-13:
Just did another test, looks about the same as last time for the dev-loop case.

```
$ make clean
$ time make -j12
.....
Linking build/rusefi_test

real    1m25.655s
user    7m42.292s
sys     0m48.458s
$ touch main.cpp
$ time make -j12
Compiling main.cpp
rm -rf build/obj/*gcda
Linking build/rusefi_test

real    0m6.738s
user    0m5.884s
sys     0m0.746s
```

rusefillc, 2022-09-01:
Just a status update:
Linking is not a problem if executed on WSL

linking performance is not rusEFI problem but NTFS problem?

While Windows 11 has build in unreliable-not-X-Windows linux UI, WLS UI is still not perfect with constant clipboard issues etc

I've even tried mounting EXT4 file system as D:\ drive but https://www.paragon-software.com/us/home/linuxfs-windows/ is not functioning at all, something is hanging up make from Windows while running from under Windoes on EXT4 drive.

mck1117, 2022-09-01:
> linking performance is not rusEFI problem but NTFS problem?

this is a well known problem - NTFS is good at some things, but running a compiler (or linker) is really not one of them.

rusefillc, 2022-09-01:
```
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$ make clean
No CCACHE_DIR
rm -f ../firmware/pch/pch.h.gch/unit_tests
Cleaning
rm -fR .dep build
Done
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$ time make -j20 2> b > a

real    1m12.118s
user    11m10.350s
sys     1m7.048s
```

rusefillc, 2025-02-27:
hello from 2025. 

``In many Dell laptops, BD PROCHOT can be triggered by a sensor in the power adapter. `` https://www.reddit.com/r/overclocking/comments/ls3yqo/help_with_understanding_bd_prochot/

throttlestop is the tool to clear

laptop#3 "xeon" , wsl

```
rusefi@DESKTOP-M056NV6:~/rusefi/unit_tests$ make clean
Cleaning PCH
rm -f ../firmware/pch/pch.h.gch/unit_tests
Cleaning
rm -fR .dep build
Done
rusefi@DESKTOP-M056NV6:~/rusefi/unit_tests$ time make -j16 2>b > a

real    2m1.584s
user    24m57.157s
sys     2m27.086s
```

rusefillc, 2025-02-27:
laptop#2 "april"

```
root@april:~/rusefi/unit_tests# make clean
Cleaning PCH
rm -f ../firmware/pch/pch.h.gch/unit_tests
Cleaning
rm -fR .dep build
Done
root@april:~/rusefi/unit_tests# time make -j16 2>b > a

real    3m58.708s
user    36m26.514s
sys     4m0.600s
```

rusefillc, 2025-03-01:
laptop#1 "primary"

[image]