rusefi/rusefi #2629: Temperature sensors issue

VasiliyKh, 2021-05-06:
Hello! This is my first issue report, so, please, don't be angry if I do something wrong.. Thank you!

There are two problems:
1. Some temperature sensors' resistance increases then temperature rises. In that case, if we put some sensors' resistance values in the TS, Frankenso gives a critical error. Is it possible to make Frankenso more friendly to this type of sensors?

2. Frankenso ignores an IAT sensor...
When I attached my IAT to INP1, the "raw IAT" gauge always whows 0 volts..., but in the real life, the voltage on the PC1 changes when I heat or cold the sensor.. Then I attached the sensor to INP8 and "raw IAT" is still the same but the voltage on PA3 changes every time I heat or cold the sensor. If I vice versa versa temperature sensors pins in the TS (the Frankenso takes data from the IAT and thinks that it is CLT, the same with IAT input). I can heat and cool the IAT and the CLT gauge reacts so good and fast, but the CLT (Frankenso thought it was IAT) is ignored.
Look like this is the software bug...

Is it possible to fix those problems?
Thank you

abelom, 2021-05-06:
Please have two separate issues for two separate problems. Let's focus on "Some temperature sensors' resistance"

1) please post a URL of your tune on rusefi.com/online
2) please post exact error message

VasiliyKh, 2021-05-06:
No problems:
[image]

Here is the tune:
https://rusefi.com/online/view.php?msq=514

rusefillc, 2021-05-06:
I've made a change, build_server should include fresh binaries in about 20 minutes from now

closed 2021-05-06 by rusefillc

VasiliyKh, 2021-05-06:
I've installed the latest version of software, I don't have any errors but the ECU still ignores the IAT sensor..

rusefillc, 2021-05-06:
one problem one ticket. i did not even read iat problem statement since separate issue

VasiliyKh, 2021-05-06:
Got it, thank you. I've just posted new issue

commit e2c5a13d (2021-05-06): Temperature sensors issue fix #2629