rusefi/rusefi #2632: IAT input is being totally ignored by Frankenso

VasiliyKh, 2021-05-06:
Frankenso ignores an IAT sensor...
When I attached my IAT to INP1, the "raw IAT" gauge always whows 0 volts..., but in the real life, the voltage on the PC1 changes when I heat or cold the sensor.. Then I attached the sensor to INP8 and "raw IAT" is still the same but the voltage on PA3 changes every time I heat or cold the sensor. If I vice versa versa temperature sensors pins in the TS (the Frankenso takes data from the IAT and thinks that it is CLT, the same with IAT input). I can heat and cool the IAT and the CLT gauge reacts so good and fast, but the CLT (Frankenso thought it was IAT) is ignored.
Look like this is the software bug...

Can we solve it somehow?
Thank you

rusefillc, 2021-05-06:
please paste 'tempinfo' and 'analoginfo' console commands output

VasiliyKh, 2021-05-06:
Do you mean open the console, put "tempinfo" and send what it'll show me, right?

rusefillc, 2021-05-06:
yes, and 'analoginfo'

VasiliyKh, 2021-05-06:
What debug mode is better to use? 
Now I have CAN-bus in debug mode

rusefillc, 2021-05-06:
debug mode and weather on Mars are not relevant for 'tempinfo' and 'analoginfo' console commands output

VasiliyKh, 2021-05-06:
Thank you for information

This is for tempinfo
[image]

This is for analoginfo
[image]

Don't look at TPSs (all throttles are disconnected now)

rusefillc, 2021-05-06:
@mck1117 do we expect ThermistorFunc or anyone to print A, B and C?

we have one command showing zero IAT voltage while another shows non-zero voltage. is that some error handling being vague?

rusefillc, 2021-05-06:
@VasiliyKh a link to tune would be nice

mck1117, 2021-05-06:
Yes, please link a tune. I would like to repro on my hardware.

VasiliyKh, 2021-05-06:
No problem, here it is
https://rusefi.com/online/view.php?msq=515

I have to admit, if I put that this is a CLT sensor (not an IAT) in the TS, but don't change anything in the hardware, the Frankenso works good with values from the sensor.  So, when it thinks that values from the IAT are values from the CLT, everything is great
It means that the problem is in the software only.. (according to my logic)

rusefillc, 2021-05-09:
@mck1117 says "Matthew Kennedy  1:05 AM
@Kavabanga found your problem with IAT.  Congrats, you are the WORLD RECORD HOLDER for "most number of analog inputs used on rusefi"

this reminds me of the external ADC dreams

FYI @andreika-git

PR #2660 (2021-05-09): move ADC pin initialization
- Move pin init to adc subscription, progress towards #2636
- Increase number of available ADC subscription entries, fixes #2632
- Fatal error if we run out of ADC subscription entries would have helped diagnose #2632
closed 2021-05-09 by rusefillc