rusefi/rusefi #2634: What exactly happens when we change connector.yaml file?

rusefillc, 2021-05-06:
I belive that gen_config goes via "cache hit, do nothing" track when we push a change to any of the connector.yaml files

we definitely do not get updated .ini in git

Do we get updated .ini in bundles?

chuckwagoncomputing, 2021-05-06:
Is there a list of files that are checked for changes? Or how does it work?
I would think that the .ini wouldn't be updated in the bundle either.

rusefillc, 2021-05-06:
```
    private static boolean isNeedToUpdateTsFiles(String tsPath, String cachePath, String cacheZipFile, List<String> inputAllFiles) {
        boolean needToUpdateTsFiles = false;
        if (tsPath != null) {
            SystemOut.println("Check the input/output TS files:");
            needToUpdateTsFiles = CachingStrategy.checkIfOutputFilesAreOutdated(inputAllFiles, cachePath, cacheZipFile);
        }
        return needToUpdateTsFiles;
    }
```

rusefillc, 2021-05-06:
```
        List<String> inputAllFiles = new ArrayList<>(inputFiles);
        if (tsPath != null) {
            // used to update .ini files
            inputAllFiles.add(TSProjectConsumer.getTsFileInputName(tsPath));
        }
```

rusefillc, 2021-05-06:
a good sign would be if/when we have relevant files included into cache.zip

rusefillc, 2021-05-10:
@DonaldBecker yes I poked and got great results

closed 2021-05-10 by rusefillc

chuckwagoncomputing, 2021-05-10:
Did you just poke the wrong person?

rusefillc, 2021-05-10:
@chuckwagoncomputing yes i did :)