rusefi/rusefi #2674: Jenkins HW CI is broken

rusefillc, 2021-05-09:
between lua merge and compile version bump and weather on Mars we get

```
Saving to: 'temp/snapshot_20210509_192315_stm32f407_rusefi/drivers/silent_st_drivers.exe'

     0K .......... .......... .......... .......... ..........  0%  613K 9s
    50K .......... .......... .......... .......... ..........  1% 1.33M 6s
   100K .......... .......... .......... .......... ..........  2% 2.03M 5s
   150K .......... .......... .......... .......... ..........  3% 2.91M 4s
   200K .......... .......... .......... .......... ..........  4% 3.52M 3s
   250K .......... .......... .......... .......... ..........  5% 4.13M 3s
   300K .......... .......... .......... .......... ..........  6% 4.68M 3s
   350K .......... .......... .......... .......... ..........  7% 4.64M 3s
   400K .......... .......... .......... .......... ..........  8% 6.58M 2s
   450K .......... .......... .......... .......... ..........  9% 5.82M 2s
   500K .......... .......... .......... .......... .......... 10% 5.29M 2s
   550K .......... .......... .......... .......... .......... 11% 4.24M 2s
   600K .......... .......... .......... .......... .......... 12% 6.52M 2s
   650K .......... .......... .......... .......... .......... 13% 6.40M 2s
   700K .......... .......... .......... .......... .......... 14% 7.79M 2s
   750K .......... .......... .......... .......... .......... 15% 3.72M 2s
   800K .......... .......... .......... .......... .......... 16% 16.6M 1s
   850K .......... .......... .......... .......... .......... 17% 16.3M 1s
   900K .......... .......... .......... .......... .......... 18% 15.4M 1s
   950K .......... .......... .......... .......... .......... 18% 13.8M 1s
  1000K .......... .......... .......... .......... .......... 19% 19.7M 1s
  1050K .......... .......... .......... .......... .......... 20% 15.2M 1s
  1100K .......... .......... .......... .......... .......... 21% 16.0M 1s
  1150K .......... .......... .......... .......... .......... 22% 6.94M 1s
  1200K .......... .......... .......... .......... .......... 23% 16.5M 1s
  1250K .......... .......... .......... .......... .......... 24% 17.4M 1s
  1300K .......... .......... .......... .......... .......... 25% 15.8M 1s
  1350K .......... .......... .......... .......... .......... 26% 15.1M 1s
  1400K .......... .......... .......... .......... .......... 27% 17.9M 1s
  1450K .......... .......... .......... .......... .......... 28% 14.2M 1s
  1500K .......... .......... .......... .......... .......... 29% 18.0M 1s
  1550K .......... .......... .......... .......... .......... 30% 13.6M 1s
  1600K .......... .......... .......... .......... .......... 31% 18.6M 1s
  1650K .......... .......... .......... .......... .......... 32% 23.6M 1s
  1700K .......... .......... .......... .......... .......... 33% 5.59M 1s
  1750K .......... .......... .......... .......... .......... 34% 21.6M 1s
  1800K .......... .......... .......... .......... .......... 35% 29.3M 1s
  1850K .......... .......... .......... .......... .......... 36% 35.0M 1s
  1900K .......... .......... .......... .......... .......... 36% 36.6M 1s
  1950K .......... .......... .......... .......... .......... 37% 26.3M 1s
  2000K .......... .......... .......... .......... .......... 38% 36.5M 1s
  2050K .......... .......... .......... .......... .......... 39% 30.4M 1s
  2100K .......... .......... .......... .......... .......... 40% 32.8M 1s
  2150K .......... .......... .......... .......... .......... 41% 3.02M 1s
  2200K .......... .......... .......... .......... .......... 42% 27.1M 1s
  2250K .......... .......... .......... .......... .......... 43% 33.3M 0s
  2300K .......... .......... .......... .......... .......... 44% 26.0M 0s
  2350K .......... .......... .......... .......... .......... 45% 23.9M 0s
  2400K .......... .......... .......... .......... .......... 46% 34.2M 0s
  2450K .......... .......... .......... .......... .......... 47% 33.2M 0s
  2500K .......... .......... .......... .......... .......... 48% 34.0M 0s
  2550K .......... .......... .......... .......... .......... 49% 32.2M 0s
  2600K .......... .......... .......... .......... .......... 50% 3.94M 0s
  2650K .......... .......... .......... .......... .......... 51% 26.9M 0s
  2700K .......... .......... .......... .......... .......... 52% 29.9M 0s
  2750K .......... .......... .......... .......... .......... 53% 25.0M 0s
  2800K .......... .......... .......... .......... .......... 54% 34.3M 0s
  2850K .......... .......... .......... .......... .......... 54% 31.5M 0s
  2900K .......... .......... .......... .......... .......... 55% 35.6M 0s
  2950K .......... .......... .......... .......... .......... 56% 32.3M 0s
  3000K .......... .......... .......... .......... .......... 57% 34.4M 0s
  3050K .......... .......... .......... .......... .......... 58% 36.5M 0s
  3100K .......... .......... .......... .......... .......... 59% 4.40M 0s
  3150K .......... .......... .......... .......... .......... 60% 22.4M 0s
  3200K .......... .......... .......... .......... .......... 61% 32.8M 0s
[... 904 lines cut ...]
    [junit] I 210509 192512.716 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] I 210509 192512.732 [main] IoUtil - Sending command [disable trigger_hw_input]
    [junit] ------------- ---------------- ---------------
    [junit] 
    [junit] Testcase: testSachs took 28,104 sec
    [junit] 	Caused an ERROR
    [junit] Was waiting for confirmation of enable pwm but got confirmation for enable self_stimulation
    [junit] java.lang.IllegalStateException: Was waiting for confirmation of enable pwm but got confirmation for enable self_stimulation
    [junit] 	at com.rusefi.io.CommandQueue.lambda$sendCommand$0(CommandQueue.java:82)
    [junit] 	at com.rusefi.io.CommandQueue.lambda$write$2(CommandQueue.java:172)
    [junit] 	at java.util.ArrayList.forEach(ArrayList.java:1257)
    [junit] 	at com.rusefi.io.CommandQueue.write(CommandQueue.java:172)
    [junit] 	at com.rusefi.io.CommandQueue.write(CommandQueue.java:159)
    [junit] 	at com.rusefi.IoUtil.sendCommand(IoUtil.java:53)
    [junit] 	at com.rusefi.functional_tests.EcuTestHelper.sendCommand(EcuTestHelper.java:101)
    [junit] 	at com.rusefi.functional_tests.EcuTestHelper.sendCommand(EcuTestHelper.java:96)
    [junit] 	at com.rusefi.functional_tests.EcuTestHelper.setEngineType(EcuTestHelper.java:136)
    [junit] 	at com.rusefi.f4discovery.CommonFunctionalTest.testSachs(CommonFunctionalTest.java:194)
```

rusefillc, 2021-05-09:
revert https://github.com/rusefi/rusefi/commit/540bb8e327027a205d00badbc1a859f5c124e2a5 did not help

now reverting compiler version update

rusefillc, 2021-05-09:
Revered from "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major" to "C:\Program Files (x86)\GNU Tools ARM Embedded\9 2019-q4-major" and windows HW CI works again.

rusefillc, 2021-05-09:
@mck1117 please try gcc 10 on your HW CI

rusefillc, 2021-09-26:
We now have 10.3-2021.07 we shall try it

rusefillc, 2022-09-28:
See #4007 to see how it fails CI

rusefillc, 2022-09-28:
Actually on the one hand we have #4584 user compaint while on the other hand we have HW CI working fine at https://github.com/rusefi/rusefi/tree/fresh-10

mck1117, 2022-10-08:
> we have HW CI working fine

hardware CI machines do not use downloaded compiler, they require a human to switch installed compiler version