rusefi/rusefi #2675: does 1mb f7 work yet

rusefillc, 2021-05-10:
flash banks? are we covered? some magic option bit? how to validate? how to set?

mck1117, 2021-05-10:
> flash banks?

Some STM32, including the STM32F7, have two independent flash banks.

> some magic option bit?

This feature is enabled or disabled with a bit in the option bytes, called `nDBANK`.

> how to validate?

In theory, the firmware now validates that the combination of `nDBANK` and device flash size are compatible.  A little bit more validation is required, though.

> how to set?

I do it in the "option bytes" page of STM32CubeProgrammer.

mck1117, 2021-05-10:
See ST AN4826: https://www.st.com/resource/en/application_note/dm00266999-stm32f7-series-flash-memory-dual-bank-mode-stmicroelectronics.pdf

closed 2021-05-10 by rusefillc

rusefillc, 2021-05-10:
[image]

not good
[image]

mck1117, 2021-05-10:
[image]

rusefillc, 2021-05-10:
[image]

PR #2678 (2021-05-10): always resolve flash address
QC builds didn't fail because they never actually resolve the flash address, which is where the check for a misconfigured device happens

fix #2675