rusefi/rusefi #2679: Proteus F7 0.6 does not start

rusefillc, 2021-05-10:
latest bundle snapshot_20210510_154441_proteus_f7_rusefi
boards passed initial QC with QC binary
red LED on start
USB detected but unable to connect

[image]

closed 2021-05-10 by rusefillc

rusefillc, 2021-05-10:
```
2021-05-10_19_05_08_793: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_011: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_121: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_345: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_563: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_779: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
2021-05-10_19_05_09_999: EngineState: CRITICAL error: 1MB single bank MCU detected: please clear nDBANK option bit and reinstall FW. 20210509@proteus
```

mck1117, 2021-05-10:
good - that means the detection works