rusefi/rusefi #2693: lua: override airmass/fuel math

mck1117, 2021-05-14:
wouldn't it be neat if you could implement a new fuel mode in Lua (calling @blundar)

blundar, 2021-05-14:
the first thought I had was to implement "familiar" (i.e. Ford, GM, maybe Chrysler?) injector models to allow "commodity" injector data published by nicer injector manufacturers to be able to be used for rusEFI.  Injector driver hardware is different which means that it won't be PERFECT but it would be a big leg up.

blundar, 2021-05-14:
but airmass, sure.... I guess I could work on that.  We'll chat

mck1117, 2021-05-14:
the challenge with implementing the injector model in lua is that injector model math happens at injection time, in an interrupt context where we have much less time to run a VM

blundar, 2021-05-14:
that seems like a really bad design choice.  once the injector math is reduced to a desired ms (or desired peak ms + hold ms, or desired phases of ms for GDI) then the interrupt routine can do whatever it needs to for generating the physical output from the desired mS, which can be computed outside of time critical interrupt space?

Same deal with ignition...   if you know desired dwell and desired firing degrees, there really isn't any good cause to do any math necessary than low level scheduling INSIDE interrupt space.

mck1117, 2021-05-14:
The reason the mass -> duration conversion happens so late is so that we can do wall wetting enrichment.  The wall wetting math adjusts individual injections (independently per cylinder), so we so it right at the moment we actually schedule the injection.

blundar, 2021-05-15:
I can see how that would make things tougher. Much tougher.
If interrupt routine simply scheduled X ms pulse with Y EOI, you'd need to update the X pulsewidth for each cylinder *really* fast outside of interrupts in order to maintain WW functionality.

closed 2026-10-21 by rusefillc