rusefi/rusefi #2713: simplify cranking ignition options

mck1117, 2021-05-16:
- maybe remove useAdvanceCorrectionsForCranking
- maybe remove useSeparateAdvanceForCranking

rusefillc, 2021-05-17:
@andreika-git FYI

andreika-git, 2021-05-17:
/firmware/config/boards/proteus/prepend.txt
+#define ts_show_advanced_cranking_settings false

mck1117, 2021-05-17:
Yes of course we could just hide it, but that isn't the point. There's no good reason I can think of to enable ignition corrections (Which could easily be +-20 degrees) while cranking.

andreika-git, 2021-05-17:
When starting a very cold engine, the fuel mixture is extremely rich, besides the gasoline evaporates slowly at low temperatures, and thus more liquid gasoline comes into the combustion chamber. Liquid fuel burns really slowly, so more early ignition is needed to burn it completely. Another reason is that when using separate advance tables for cranking/idling, there can be a significant difference when transitioning from cranking curve to idle (due to corrections disabled for cranking), which leads to noticeable engine wobbling at cranking rpm threshold.

As for the "huge 20 degree" corrections, here is the example of CLT correction curve tested on at least 2 different real cars (in winter) and currently used:
[image]
Even at -40C it adds only 8 degrees. If someone has a different curve with more aggressive angles, then... well, just don't enable the correction for cranking.
The second one, the IAT correction, usually doesn't affect the cold start at all:
[image]
The third fsioTimingAdjustment correction is very user-specific and cannot be taken into consideration.
Finally, the last pidTimingCorrection is disabled for cranking anyway (if (phase != Phase::Idling) return 0).

mck1117, 2021-05-17:
> so more early ignition is needed to burn it completely

OEM tunes do not necessarily reflect this:

[image]

Top two tables control open loop running ECT correction.  Bottom table is cranking.  Essentially no change to cranking advance, except when very hot.

> The second one, the IAT correction, usually doesn't affect the cold start at all:

That's not the default table.  Default table has +4 degrees in the bottom left corner, and 3 degrees at 10 C.  You could very easily end up with an accidental extra 8-12 degrees of timing, which on my engine is enough to get starter kickback.

blundar, 2021-05-17:
Cranking is its own beast.  Closed loop spark corrections appropriate for idle or general operation would not be good in the least.  There is a very good reason that crank operation hands off to run operation at a pretty low RPM, quite a bit lower in many OE situations that we do.
Cranking spark vs, ECT = good.
Using the same timing scale as normal operation... meh

blundar, 2021-05-17:
Granted much is IMHO, but its a method I've honed over decades.
https://rusefi.com/forum/viewtopic.php?f=16&t=2021

mck1117, 2021-05-17:
>  Closed loop spark corrections

he's right here, this one does actually disable itself while cranking