rusefi/rusefi #2721: Something seems wrong with H121 vag pinout

rusefillc, 2021-05-20:
We had a recent change in 121 yaml and no nothing gets added into the table

I assume something is not great with the yaml? Can we please fail with a more explicit error message if that's the case?

https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/hellen/hellen121vag/connectors
https://rusefi.com/docs/pinouts/hellen/hellen121vag/

chuckwagoncomputing, 2021-05-20:
I see the problem. I should have time to fix it this afternoon.

chuckwagoncomputing, 2021-05-20:
Also @BeerMoneyMotorsports there is a tiny problem with the yaml. A pin is listed as "1a" but the coordinates are for "1A".

BeerMoneyMotorsports, 2021-05-20:
@chuckwagoncomputing i havent gotten as far as the coordinates yet.  i will make sure those are consistent with each other whenever i get to that point.

rusefillc, 2021-05-20:
fixed thank you!

closed 2021-05-20 by rusefillc