rusefi/rusefi #2732: Timing test mode

rusefillc, 2021-05-23:
From does not have high voltage wires

What kind of helpers can we add? Map signal analysts? Noid light? Else?

mck1117, 2021-05-23:
what?

OrchardPerformance, 2021-05-23:
Copying the comment from slack: 

Is this another argument for having a coded in "timing light" function that when enabled allows you to set and extra pin the pulses at the same time as Ign1 output?
Thus allowing you to trigger a timing light or dumb flash light via transistor.
I know it's coding someone would have to do but it seems like a sensible thing to have given the difficulty around timing lights and COPs
I feel that is something that comes under the "smart ECU"/better ecu than others.

rusefillc, 2021-05-23:
Timing led is a possibility

rusefillc, 2021-05-23:
@OrchardPerformance @mck1117 I wonder if we should simply document how to use existing tachometer output as a timing LED? looks like timing LED is just a special case with "events per engine cycle" == 1

rusefillc, 2021-05-23:
@OrchardPerformance @andreika-git but another direction is better TDC position validation

can we validate TDC against MAP sensor? against MAF sensor? anything else to help user tell his that his TDC value is more wrong than not?