rusefi/rusefi #2733: cranking timing not costant

rusefillc, 2021-05-23:
[image]

rusefillc, 2021-05-23:
tune https://rusefi.com/online/view.php?msq=531
log https://rusefi.com/online/view.php?log=235

rusefillc, 2021-05-23:
i do not like minCrankingRpm static variable without a reset etc

rusefillc, 2021-05-23:
@andreika-git do we need a check box to disable taper between cranking threshold and current RPM?

this advanced logic works really bad if all your angles are very very wrong, there should be a way to run "simplest" configuration on demand

mck1117, 2021-05-23:
setting running mode to "fixed" will also give predictable results

andreika-git, 2021-05-23:
> do we need a check box to disable taper between cranking threshold and current RPM?

We already have a way to disable taper if we really need to - "Use separate Advance Table for cranking". Not the best way possible but it allows to set the fixed value for cranking (if the table if filled with a constant value)...

rusefillc, 2021-05-23:
Problem is that _no one_ knows about the taper. There is no way to know to use table if plain value is needed.

rusefillc, 2021-05-23:
Well, maybe this could be improved by improving only the TS file not changing firmware

See also #2735

rusefillc, 2021-05-23:
[image]

[image]

closed 2021-05-23 by rusefillc