rusefi/rusefi #2739: Support dual fans

mck1117, 2021-05-25:
Two fan outputs, each should support (independent):
- on/off tempersture for hysteresis
- Enable this fan with AC

dron0gus, 2021-05-25:
Some old JDM cars have two two-speed fans. Controlled through two signals and some relay magic. :)
[image]

mck1117, 2021-05-25:
Yep - similar on the C5 corvette. One signal turns on both fans in series (half voltage), and the other switches some relay magic to put them in parallel instead.

BeerMoneyMotorsports, 2021-06-10:
i like this for miata specifically.  main cooling fan for normal function.  second cooling fan activated if temps get higher OR AC is on.  Hmm, maybe even have both fans on with AC.

stefanst, 2021-06-10:
You don't really need the big fan if the engine is cool-ish and the AC is running. Once things get a little hotter, the big fan kicks in anyway.

mck1117, 2021-06-10:
yeah, not as necessary on the miata, but some cars do have two fans next to each other, and run them in a high/low speed arrangement

PR #2806 (2021-06-13): dual fans
enable second fan

fixes #2739 

depends on #2805
closed 2021-06-13 by rusefillc