rusefi/rusefi #2768: Idle auto and manual flipped or broken

rusefillc, 2021-05-30:
Auto idle air does not seem to work
Manual seems to apply pid regulator
Do we have bugs or project options are flipped?

How on eaeth have we not noticed?!

rusefillc, 2021-05-30:
https://youtu.be/OamrTJtsiHo
English voice in the end

mck1117, 2021-05-30:
do you have a link to a tune and log?

rusefillc, 2021-05-30:
https://rusefi.com/online/view.php?msq=540
https://rusefi.com/online/view.php?log=250
https://rusefi.com/online/view.php?log=249

mck1117, 2021-05-31:
Notice our friendly neighborhood coasting/idling indicators.  If it says coasting while you say it should be idling, then you need to fix that 🙂 

[image]

mck1117, 2021-05-31:
WTF is this?

This is what you were seeing in manual mode.  Small CLT changes between 88-100 degrees will cause huge changes in idle position.  Fix your tune, dude!

[image]

rusefillc, 2021-05-31:
while testing we've tried pretty weird numbers and we were not getting the response we were expecting

[image]

mck1117, 2021-05-31:
what do you mean? Math adds up to me.

At timestamp 2:31 in the video:
CLT = 93.xx
base idle = 19

Manual idle mode really is as simple as base * table(clt).

At 93.5 deg, you'll end up with a multiplier of 1.5.

1.5 * 19 = 28.5, which is exactly what the position gauge shows.

rusefillc, 2021-05-31:
I guess let's start from the beginning

We are well aware of idle/coasting indicators
Dron and myself were unable to see auto idle showing any signs of life while we have tried

We went all to P of 5 yet no movement on indicator.

Maybe we have a major usability issue with auto conditions

mck1117, 2021-05-31:
Manual mode:
When you mention "PID regulation" as the idle position is moving at a few % per second, notice that the CLT is also dropping pretty rapidly, sliding you up the right side of the huge bump in the table.
Auto mode:
> Maybe we have a major usability issue with auto conditions

yes, see #2745 :slightly_smiling_face:

Specifically, I think your problem is here (see linked comment on other PR): https://github.com/rusefi/rusefi/pull/2745/files#r642107332

mck1117, 2021-05-31:
> Maybe we have a major usability issue with auto conditions

To expand, yes, I agree, if the green "IDLING" indicator is lit up, I would expect to see position moving around to maintain idle speed.  Any condition where that doesn't happen is a bug.  #2745 specifically fixes that bug.

rusefillc, 2021-05-31:
at the moment the PRs are not minimal fixed focusing on the root cause of the issue

I am not comfortable with refactoring, logic changes and fixed are piled together. we need to split into those three explicit categories.

mck1117, 2021-05-31:
Yes, see #2776.  That PR carves the non-logic changes off of #2745.

mck1117, 2021-05-31:
And #2779 for a related fix.

rusefillc, 2021-05-31:
my understanding of healthy SDLC tells me that
step 1: automated code coverage has to be merged into master to prove the bug
step 2: minimal bugfix with code coverage change would need to be merged next

I am extremely uncomfortable of piling refactoring with bugfixing and maybe unrelated logic changes with not enough automation

it's 2021 and we have a major bug in a crucial functionality. How did we get here and how do we change the SDLC to reduce the chance of being here again?

rusefillc, 2021-06-14:
@mck1117 close or not?

mck1117, 2021-06-14:
I think close!

closed 2021-06-14 by rusefillc