rusefi/rusefi #2770: Fuel Source for cranking triggers "please power cycle" warning

rusefillc, 2021-05-30:
that's unexpected

mck1117, 2021-05-30:
ctrl-f of that string yields nothing, what is the string in the warning?

mck1117, 2021-06-01:
poke?

rusefillc, 2021-06-02:
discovery with a week old firmware, week old project
change "fuel source for cranking"
got
[image]

mck1117, 2021-06-03:
No cranking settings are marked with `requiresPowerCycle`, so I'm not sure what you saw.

rusefillc, 2021-06-03:
I know! Yet @dron0gus and myself say it at the garage and the image above is me reproducing this with a discovery in 10 seconds

mck1117, 2021-06-03:
no way it repros for me too connected to simulator

rusefillc, 2021-06-03:
Welcome to IT, or probably more like welcome to TS

mck1117, 2021-11-29:
moved to ts-issues repo

closed 2021-11-29 by mck1117

LT401vette, 2021-11-30:
Power cycle required is monitored at the defined data type level, in this case a U32, not bit level.  useRunningMathForCranking is a bit field sharing that U32 with etb_use_two_wires which is set to requiresPowerCycle.
I will evaluate the impacts of monitoring at the bit level.

LT401vette, 2021-11-30:
Making this byte level should not be difficult, Taking it to the bit level would require changes to the framework. That would have greater impacts. Certainly not something to be done in a patch release.

LT401vette, 2021-11-30:
After a thought, I came up with another way to track the bit fields. Implemented for next beta.