rusefi/rusefi #2771: Maybe split advanced cranking settings

rusefillc, 2021-05-30:
Nobody cares about cranking dwell time shall we hide it from current polluted dialog?

stefanst, 2021-05-30:
Clarify statement?
Cranking dwell is usually longer than running dwell.

rusefillc, 2021-05-30:
Yes you are right

Point is different: at the moment we mix options which require custom or often affection with 6ms cranking idle which might be universally OK for everyone this should be more bed aside to refuse clutter

mck1117, 2021-05-31:
we *absolutely* care about cranking dwell time.

We *do not* care about cranking dwell angle mode, so we can hide that a little better.

mck1117, 2021-06-03:
@rusefillc is this now done now that it looks like the 3 boxes on the right?

[image]

closed 2021-06-03 by rusefillc