rusefi/rusefi #2799: inhibit starting without oil pressure

mck1117, 2021-06-08:
I may or may not have started an engine this weekend without any oil in it.  We have software, let's prevent that!

rusefillc, 2021-06-08:
I like it

mck1117, 2021-06-08:
I think the right thing to do is have two parameters:
"minimum starting oil pressure"
and
"minimum running oil pressure"
There are a few corners at The Ridge Motorsports Park that we can hit very low oil pressure (<10 psi where there should be 50) for about a second (low load, going over a big crest in a hard left hand corner) where we don't want the engine to cut out, but it's perfectly reasonable to inhibit start without oil pressure.

rusefillc, 2021-06-08:
I think preventing from cranking is legit while we probably never want to completely shut engine off?

mck1117, 2021-06-08:
Looking at some logs of oil pressure during startup it might need to be a little smarter.  Something like "if oil pressure doesn't go above X within Y seconds of start, kill engine".  Simply waiting for oil pressure before firing could significantly increase start time, especially on a really cold engine that's been sitting for days (draining oil out of places, so you have to displace air first).

closed 2021-06-28 by mck1117

mmatousek89, 2021-07-16:
lets do it other way, start engine, and if there is no "given pressure" within "given delay" then fuelcut for "given shutdown time"

mck1117, 2021-07-16:
@mmatousek89, that's exactly what #2800 does.