rusefi/rusefi #2815: Review default H64 tune

rusefillc, 2021-06-14:
@BeerMoneyMotorsports how large are the gaps between your personal H64 tune and default H64 tune? Are there any improvements to make to the default tune?

BeerMoneyMotorsports, 2021-06-14:
this tune from the stock 95 is ok.  https://rusefi.com/online/view.php?msq=537

rusefillc, 2021-06-16:
@BeerMoneyMotorsports that's a step forward!

Can you please help my more explicitly listing all differences between current NA6 default tune and https://rusefi.com/online/view.php?msq=537?

rusefillc, 2021-06-23:
acRelayPin? acSwitch?

I am looking for help with the legwork of what's missing in the canned tune.

BeerMoneyMotorsports, 2021-06-27:
These are some screenshots of key tune differences for NA6. Right side are the correct settings.

[image]

NA6= 212cc 94-95= 254cc  http://miataturbo.wikidot.com/fuel-injectors
[image]

[image]

[image]

[image]

rusefillc, 2021-06-28:
@BeerMoneyMotorsports see https://github.com/rusefi/rusefi/commit/ec72c6950ef67445ef03e49cd77c5bc11b9cc5ab some progress

TODO 94 injectors pinout
TODO A/C and idle stuff?

BeerMoneyMotorsports, 2021-06-28:
90-95 have same pinout for injectors for sequential(standard sequential on 94-95).  the defaults for injectors did not require modification.  

Here is fan and AC setup for 90-93
[image]

Here is the fan and AC for 90-95
[image]

Idle hardware for 90-95
[image]

Idle settings that should be a solid start for most.
[image]

BeerMoneyMotorsports, 2021-06-28:
Here are the AC setup differences.  Right side is correct.   Idle seems to be fine.

94/95 AC setup
[image]

90-93
[image]

rusefillc, 2021-06-28:
@BeerMoneyMotorsports we have the following in H64 yaml:

```
  - pin: 1L
    id: GPIOD_12
    class: outputs
    ts_name: 1L - Radiator Fan 94-95
    function: Radiator Fan Relay
    type: ls
```

BeerMoneyMotorsports, 2021-06-28:
hmm, the yaml is correct. 

1L = coolant fan 94/95
1R = Coolant fan 90/93

In the H64 rev A there was no provision for 90-93 fan so it is a physical board change and uses the 94/95 output.  fixed in H64 rev B

Notice that 1L and 1R have the same ID in the yaml.  Looks like 1L does not show up in the pin list for Tuner Studio.  Only 1R

rusefillc, 2021-06-28:
Progress under https://github.com/rusefi/rusefi/commit/70bc5b2375648562386a9eb399dc42c304aa7e20

Also https://github.com/rusefi/rusefi/issues/2877 created

BeerMoneyMotorsports, 2021-06-29:
target lamba, can we implement this?  currently everything is set to 1.0

[image]

rusefillc, 2021-06-29:
yes if you give me tune URL

BeerMoneyMotorsports, 2021-06-29:
https://github.com/rusefi/rusefi/issues/2878

rusefillc, 2021-07-24:
@BeerMoneyMotorsports where are we with this ticket overall