rusefi/rusefi #2819: real dash integration

rusefillc, 2021-06-15:
Does anyone own real dash?

Jani emailed:

```
I took a look at the files, and I think it will work with one change to
the ini file:
In [OutputChannels] change:
ochGetCommand  = "O%2o%2c"
to
ochGetCommand    = "A"
If you can verify that connection works with this change to the ini
file, I can make the parsing of ochGetCommand automatic so users do not
have to modify their ini files.
```

stefanst, 2021-06-15:
real dash is free with ad support.

rusefillc, 2021-06-15:
@shadowm60 please attempt custom firmware?

shadowm60, 2021-06-17:
Should i modify and rebuild with ini file, and try to connect with realDash? if that would work, do we update the ini file from now on?

mck1117, 2021-06-17:
They said they would fix their ini parsing if we could observe it working, right?

shadowm60, 2021-06-17:
ok, i build the test code, and if all works correctly, i can identify the deltas, and try to contact realDash to make the update.

mck1117, 2021-06-17:
yeah, see if you can get realdash working, and if so, report back what it took at least

rusefillc, 2021-07-18:
%2o%2c means "two byte range offset and two byte block count/size" i.e. RE supports only the "advanced" version of the command with block range specification

it looks like real dash does not support the advanced command and instead they would want us to implement the basic "A" command to output complete gauges not only specified range

it should be totally possible to have both commands in the firmware just need someone to add the code

mck1117, 2021-07-19:
Or just use the same command, and if it arrives without offset/size parameters, assume offset 0 and full size.

mck1117, 2021-07-19:
Did we do what she said, and try just making that edit and matching change to fw?

shadowm60, 2021-07-19:
I have tried ini modification, but it did not worked, i am trying to understand what is happening and what needs to be touched to make it work

rusefillc, 2021-07-19:
please create a PR even with very broken stuff, it would allow us to look at general direction

duhafnusa4, 2021-07-23:
should somebody update this?

https://www.realdash.net/forum/viewtopic.php?f=4&t=1281&sid=01380fbb1c0abfb03c9a6a171c4e94c6

duhafnusa4, 2021-07-23:
4/30/2021 was the last time we commented on the realdash forum, maybe update?

https://www.realdash.net/forum/viewtopic.php?f=4&t=1281&p=5831&hilit=rusefi#p5831

duhafnusa4, 2021-08-31:
@rusefillc can we close this ticket as we have integration working?

https://rusefi.com/forum/viewtopic.php?f=5&t=2107&p=42031#p42031

https://www.realdash.net/forum/viewtopic.php?f=4&t=1281

duhafnusa4, 2021-08-31:
also Real Dash Beta 1.9.5 calls out rusefi support

https://www.realdash.net/forum/viewtopic.php?f=2&t=1325

closed 2021-08-31 by rusefillc