rusefi/rusefi #2828: Alex and Abricos are complaining about fuel

rusefillc, 2021-06-18:
New

Alex Miculescu  10:14 AM
@Matthew Kennedy did we changed something drastical in fuel algo?
10:15
image.png 
image.png

10:15
i have <50 VE in the table, but ecu is using > 50
10:15
car runs like crap
:+1:
1

10:15
started autotune today to make it home at least
10:16
it's like VE would not be extracted from the table at all
10:17
i have a trace with VE 7 in the table in the fuel:ve shows 64....
10:17
engine is hot, multipliers are all at 1.0
10:20
https://i.imgur.com/4fWBmMy.png

(200 kB)
https://i.imgur.com/4fWBmMy.png

10:20
looks like i cannot upload the log
10:20
the current tune is uploaded

Matthew Kennedy  1:23 PM
Do you have idle ve table enabled?

abricosvw  4:06 PM
Something really not good
4:06
I have same ve problem

Matthew Kennedy  4:18 PM
my engine runs fine
4:18
idk

Matthew Kennedy  4:30 PM
@Alex Miculescu check "engine load accel enrichment"
4:34
wow autotune has made a real mess of your VE map

Alex Miculescu  4:45 PM
it is a real mess with autotune
4:45
but it looks like it would not react to ve table
4:45
i do not have idle ve table enabled

Matthew Kennedy  4:45 PM
can you try turning off the load AE

Alex Miculescu  4:45 PM
i did auto update (dfu mode, console did not connected to the board, so i used the ST tool to connect to the board and flash)
4:46
i will try tomorrow the debugger reflash
4:46
AE is multiplier 0
4:46
taper all 1.0
[... 559 lines cut ...]
Dave B.  11:49 PM
and the FRPP cals are done by different than Ford OEM

Matthew Kennedy  11:49 PM
and the pressure tables match
11:50
low slope is different

Dave B.  11:50 PM
you realize there are 3 revs of gt500 tune
11:50
"GT500" in database (07-08) and XSH1 (09-10) and then 11-12 with tricore/copperhead

Matthew Kennedy  11:50 PM
this is a 2008

Dave B.  11:51 PM
that data will probably match XSH1
11:51
it's a quite minor difference

Matthew Kennedy  11:53 PM
it mostly matches the ford racing data

Message dev_fuel_algorithms

shadowm60, 2021-06-18:
working rel:
tune: https://rusefi.com/online/view.php?msq=560
log: https://rusefi.com/online/view.php?log=262

failed nightly:
tune: https://rusefi.com/online/view.php?msq=559
log: https://rusefi.com/online/view.php?log=263

mck1117, 2021-06-26:
I can reproduce using Alex's tune:
[image]

mck1117, 2021-06-26:
`m_veTable->getValue(rpm, load)` returns 76

mck1117, 2021-06-26:
Here's the problem:

[image]

rusefillc, 2021-06-26:
thats a swert one! lets halt and catch fire?

mck1117, 2021-06-26:
yes

mck1117, 2021-06-26:
This is a user config error, will be detected by #2866, closing

closed 2021-06-26 by mck1117