rusefi/rusefi #2839: TS Burned All Data engine stop

Abricosvw, 2021-06-20:
when TS burn itself, then the engine stops ... 
i checked 3 latest firmware releases ...
[image]

rusefillc, 2021-06-21:
I am able to reproduce this with a discovery

1) step 1: full chip erase and flash rusefi_bundle Garbage Man Day.zip
2) connect with TS
3) enable internal stimulator
4) go make any change, hit "Save"

actual result: I get blue LED freee and write
expected result: no blue LED freeze, "unsaved changes" indicator in TS

rusefillc, 2021-06-21:
I assume we've broken it while making logic smarter defect which chips have two banks and allow hot write?

mck1117, 2021-06-21:
This is expected behavior - we're supposed to burn while self stimulating

rusefillc, 2021-06-21:
Ok, now with external stim

1) I set trigger to "one tooth"
2) I have one wire from PD1 to PC6
3) I play with "displacement" field and "save" button

Sometimes things are great. Sometimes sync is lost and RPM goes to zero. At least once I got 3k or 4k for a moment instead of 1200.

rusefillc, 2021-06-21:
Relevant console commands

"ext_stimulation"
"burnconfig"

dron0gus, 2021-06-21:
Looks like scheduling issue. Something is blocking CPU for a while.
Self stimulation. Subaru SVX trigger. First trigger is 12 pulses per rotation. With equal gaps.
[image]

dron0gus, 2021-06-21:
1. enable self stimulation
2. go to High Speed Logger, start
3. open Ignition->Ignition cylinder trim
4. Adjust any value in this dialog
5. observe trigger errors.
Somehow TS calls flash write on any change on these values.

rusefillc, 2021-06-21:
self stimulation is a special case

to reproduce real issue you have to use a wire and external stimulation if on discovery 

```
	if (ENGINE(directSelfStimulation) || engine->rpmCalculator.isStopped()) {
		/**
		 * rusEfi usually runs on hardware which halts execution while writing to internal flash, so we
		 * postpone writes to until engine is stopped. Writes in case of self-stimulation are fine.
		 *
		 * todo: allow writing if 2nd bank of flash is used
		 */
#if EFI_INTERNAL_FLASH
		writeToFlashIfPending();
#endif /* EFI_INTERNAL_FLASH */
	}
```

dron0gus, 2021-06-21:
```
void writeToFlashIfPending() {
	// with a flash write thread, the schedule happens directly from
	// setNeedToWriteConfiguration, so there's nothing to do here
	if (**allowFlashWhileRunning()** || !getNeedToWriteConfiguration()) {
		return;
	}

	writeToFlashNow();
}
```
So this path does not leads to `writeToFlashNow()`

Abricosvw, 2021-06-21:
it seems that the settings are always save ...
in 50 minutes, there were no settings that were never saved ...
the engine has never stopped while burn settings (any) ...

and ( unsaved changes ) are always ON and     one time ( failed CRC ) ...

[image]
[image]

PR #2860 (2021-06-25): fix trigger changed logic
You can't compare arrays by value in c++.

Fixes #2839
closed 2021-06-25 by rusefillc