rusefi/rusefi #2857: Ant invocation in "java_console" should build java console

rusefillc, 2021-06-25:
it should take one ant invocation not two to build java console in a freshly checked out folder

whatever dependency on antlr has to be handled automatically

rusefillc, 2021-06-25:
related to https://github.com/rusefi/rusefi/commit/c0aeb39301a9248a5d043083befc553dba0c4cd1

mck1117, 2021-06-25:
I think the real question is: why does the console depend on configuration_definition at all? Can you move any stuff actually needed by both to some `shared/` project to remove the dependency completely?

rusefillc, 2021-06-25:
java_console does not have binary dependency on configuration_definition it builds same source code. I do not follow the argument.

mck1117, 2021-06-25:
Why does the console build all of the code in `configuration_definition/`? Surely the console doesn't need _all_ of the code in `configuration_definition/`.

rusefillc, 2021-06-25:
Because we balance costs with benefits. While we have a zillion of things to improve it might make most sense to approach those zillion things in the order of subjective priority.

rusefillc, 2021-08-31:
@dron0gus says Hi

rusefillc, 2021-11-07:
@mck1117 every time we change grammar I have to do boring stuff to work around locally :(

Yes proper fix is gradle migration

rusefillc, 2021-12-22:
@alxrMironovHome here is the work around;

cd java_tools
cd configuration_definition
ant

once you invoke 'ant' in configuration_definition you should be able to invoke 'ant' in java_console

rusefillc, 2022-07-22:
"Not cool when we introduce extra manual labor with no attempt to undo. That's not progress." would be my passive-aggressive take on https://github.com/rusefi/rusefi/pull/4365#issue-1313607538

rusefillc, 2022-07-22:
Anyway #4371

closed 2022-11-01 by rusefillc