rusefi/rusefi #2861: VW warnings on cranking

Abricosvw, 2021-06-25:
9002 

CUSTOM_SYNC_COUNT_MISMATCH = 9002

Abricosvw, 2021-06-25:
https://rusefi.com/online/view.php?log=270

Abricosvw, 2021-06-26:
[2021-06-25_19.58.35.zip](https://github.com/rusefi/rusefi/files/6719293/2021-06-25_19.58.35.zip)

mck1117, 2021-06-27:
It's seeing an early sync point here:

[image]

mck1117, 2021-06-27:
Then a real one here:
[image]

mck1117, 2021-06-27:
@Abricosvw was this log made with a MAX992x chip on Frankenso?

If so, swap your wires:

[image]

mck1117, 2021-06-27:
https://github.com/rusefi/rusefi/wiki/Trigger-Configuration-Guide

rusefillc, 2021-06-27:
@mck1117 you are right about the early sync point but making gap expectation smarter does not improve things

```
	s->setTriggerSynchronizationGap2(1.6, 4);
	s->setSecondTriggerSynchronizationGap(1); // redundant gap ratio to fight start-up noise
	s->setThirdTriggerSynchronizationGap(1);
```

the warning is about constant 
```
sync point: isDecodingError=1
count: cur=116 exp=58
count: cur=0 exp=0
count: cur=0 exp=0
```
which I believe is related to a bug in expected tooth calculaton

mck1117, 2021-06-27:
That warning is a product of the broken csv replay logic :)

rusefillc, 2021-06-27:
@mck1117 any specific patched you would want to contribute? :)

mck1117, 2021-06-27:
crafting a PR now!

mck1117, 2021-06-27:
with #2874, tracking two extra gaps /does/ prevent the warning!
```
s->setTriggerSynchronizationGap2(1.6, 4);
s->setSecondTriggerSynchronizationGap(1); // redundant gap ratio to fight start-up noise
s->setThirdTriggerSynchronizationGap(1);
```

Abricosvw, 2021-06-27:
Yes I have MAX chip on Frankenso ...
This my trigger (pic from internet)
[image]
[image]

rusefillc, 2021-06-27:
Trigger decoder improved for 60/2 VW (only that version for now)

rusefillc, 2021-07-04:
[image]

[image]

rusefillc, 2021-07-04:
TODO: download log files check what other warning codes are there

i believe there is CAM configured without actual signal?

rusefillc, 2021-07-04:
by the way #2898

Abricosvw, 2021-07-04:
when i swap VR pin

[trigger 36.1 salea.zip](https://github.com/rusefi/rusefi/files/6760060/trigger.36.1.salea.zip)

rusefillc, 2021-07-07:
closing this one just to reduce confusion

mck1117, 2022-02-16:
I think we've fixed this, closing.

closed 2022-02-16 by mck1117