rusefi/rusefi #2863: Smart ECU: injector diagnostics

rusefillc, 2021-06-26:
Can we do anything to diagnose injectors without taking engine apart?

rusefillc, 2021-06-26:
Maybe add 10% of extra fuel into each injector one by one and validate that AFR has changed more or less equally regardless of which injector was floating extra fuel? Does this make _any_ sense?

JamesBalazs, 2021-07-06:
Opcom/Tech2 offers live injector deviation measurements on some Vauxhall engines (Z19DTH controlled by EDC16).

I believe they do this by comparing the change in crank speed caused by each combustion event. They calculate an arbitrary value for each cylinder, with an acceptable range of about +/- 3.0

At idle this gives a pretty good indication of leaking or failing injectors. Just an idea.