rusefi/rusefi #2877: H64 uses GPIOD_12 twice why does it not fail?

rusefillc, 2021-06-28:
https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/hellen/hellen64_miataNA6_94/connectors/main.yaml uses GPIOD_12 for both pin: 1L and pin: 1R

@chuckwagoncomputing in my opinion it should fail?

rusefillc, 2021-06-28:
@BeerMoneyMotorsports FYI

chuckwagoncomputing, 2021-07-02:
I don't know Java too good.

https://github.com/rusefi/rusefi/blob/53e4908497b8d652bd539804d2d439be7860e31a/java_tools/configuration_definition/src/com/rusefi/ConfigDefinition.java#L456

rusefillc, 2021-07-02:
@chuckwagoncomputing so https://www.geeksforgeeks.org/difference-equals-method-java/ ?

rusefillc, 2021-07-02:
@chuckwagoncomputing https://github.com/rusefi/rusefi/commit/a27edf50fb17a2ecbf7f4deeb20d27ab1aaeef4c

rusefillc, 2021-07-02:
@chuckwagoncomputing thank you, this now fails as expected!

@BeerMoneyMotorsports we now have scripts detecting the issue in H64, which section is using GPIOF_12 rightfully which one does not?

```
  - pin: 3I
    id: GPIOF_12
    class: outputs
    ts_name: 3I - Tachometer 90-95.5
    function: tachometer
```
vs
```
  - pin: 1F
    id: GPIOF_12
    class: outputs
    ts_name: 1F - Tachometer 95.5
    function: Tachometer
    type: ls
```

closed 2021-07-02 by rusefillc