rusefi/rusefi #2882: HTML title for pinout pages

rusefillc, 2021-06-30:
Shall we add a nice <title> for https://rusefi.com/docs/pinouts/hellen/hellen121nissan/ and siblings?

chuckwagoncomputing, 2021-07-24:
Now do we leave this open until yaml fields have been added, or close it now that the capability is present? I closed #3012 before I thought about that.

rusefillc, 2021-08-12:
[image]

do not follow. do you mean we need to add actual values into actual yaml files? at the moment HTML content does not have TITLE tag, is there a way to get HTML HEAD TITLE by specifying some yaml field? can you please specify that at least for one yaml file as example?

i wonder if there is some confusion here?

rusefillc, 2021-08-12:
I will take care of those, thank you!!!

commit 498f1681 (2021-08-15): HTML title for pinout pages fix #2882
closed 2021-08-15 by rusefillc