rusefi/rusefi #2893: v12 m73 either start or does not

rusefillc, 2021-07-01:
m73 with default m73 proteus tune
only rusefi car with starter button

1/3 of times starter engages for a moment and silenve
2/3 of times engine starts

rusefillc, 2021-07-01:
maybe related to #2886 ? maybe not?

mck1117, 2021-07-01:
what condition do we wait for to disable the starter?

rusefillc, 2021-07-01:
RPM > config(crankingRpm) hopefully?

mck1117, 2021-07-01:
```
	if (engine->rpmCalculator.isRunning()) {
		// turn starter off once engine is running
		bool wasStarterEngaged = enginePins.starterControl.getAndSet(0);
		if (wasStarterEngaged) {
			efiPrintf("Engine runs we can disengage the starter");
			engine->startStopStateLastPushTime = 0;
		}
	}

	if (getTimeNowNt() - engine->startStopStateLastPushTime > NT_PER_SECOND * CONFIG(startCrankingDuration)) {
		bool wasStarterEngaged = enginePins.starterControl.getAndSet(0);
		if (wasStarterEngaged) {
			efiPrintf("Cranking timeout %d seconds", CONFIG(startCrankingDuration));
			engine->startStopStateLastPushTime = 0;
		}
	}
```

mck1117, 2021-07-01:
do you see either of these messages?

rusefillc, 2021-07-02:
I was not looking :(

Next step is getting 1+ Proteus in case from @stefanst to do more testing. Overall that's a low priority for me just needed a record of the issue - ideally I would wait until I have x5 working Proteus to test on v12 :)

mck1117, 2022-02-16:
was this fixed by new spark plugs?

rusefillc, 2022-02-16:
yes it was fixed by 6 spark plugs

rusefillc, 2022-02-16:
this is also begging for "your spark plugs are wet" diagnostics message which is #3771

closed 2022-02-16 by rusefillc