rusefi/rusefi #2921: Matt has CUSTOM_ERR_VVT_OUT_OF_RANGE warning

rusefillc, 2021-07-07:
See https://rusefi.com/forum/viewtopic.php?p=41486#p41486

mck1117, 2021-07-07:
I don't understand where these values are coming from. It's bouncing between ~1309 and ~0.3.

~0 I understand: the rising/falling edge is just about exactly at TDC on my engine, modulo timing chain slop (no tensioner, pushrod engine).
Where did ~1309 come from? 1309 modulo 360 is 229.  Nothing about my engine is 229 degrees.  Where did that number come from?

mck1117, 2021-07-07:
Will try to reproduce on the bench tonight using single tooth crank, single tooth cam (both of which I can generate with an arbitrary waveform generator, and adjust the phase between them)

rusefillc, 2021-07-07:
I still have Nissan VVT jumping
and do not have time to test on NB2 :( overall I know this area is messy :)

mck1117, 2021-07-07:
well, we now know it reproduces on single tooth cam - so it's nothing special about the trigger decoder itself.  I'll test tonight and see what I find.

rusefillc, 2021-07-07:
Do we need three more tickets for the three other warnings?...

PR #2926 (2021-07-12): fix vvt indication
- VVT output channels are signed, so they can output negative values (fixes TS)
- VVT position does indeed need to get wrapped (offset isn't enough) - if the VVT sync point moves across the crank sync point, we need to smoothly output negative -> positive values, instead of wrapping all the way around.
- Normalize TS channel names: bank 1/2 intake/exhaust
- Now that we auto-wrap VVT position, remove warning about VVT out of range

fix #2755 
fix #2921
closed 2021-07-12 by rusefillc