rusefi/rusefi #2925: EFI_ADC_13 used multiple times in hellen121vag

chuckwagoncomputing, 2021-07-08:
https://github.com/rusefi/rusefi/blob/9dfc066bce71f7842ee5ac8cc8fdfc5cfb983910/firmware/config/boards/hellen/hellen121vag/connectors/main.yaml#L75

https://github.com/rusefi/rusefi/blob/9dfc066bce71f7842ee5ac8cc8fdfc5cfb983910/firmware/config/boards/hellen/hellen121vag/connectors/main.yaml#L213

Do we have a case of two pins wired to the same ECU pin? If so, I don't believe it's possible to detect pins used multiple times without false negatives. Unless we add another key in the yaml or something.

mck1117, 2021-07-08:
how is the board actually wired? are both of those pins tied together, or are they somehow different inputs?

chuckwagoncomputing, 2021-07-08:
Looks like they are actually tied together.

chuckwagoncomputing, 2021-07-13:
Well, this and #2897 are in conflict. I suggest we close both, and not worry about detecting duplicates. The alternative would be to have a field that says "ignore this duplicate".

closed 2021-07-13 by rusefillc