rusefi/rusefi #2931: trailing spark outputs

mck1117, 2021-07-08:
Fire a second coil, delayed from the first by a programmable angle (table). Used for Mazda Rotary, engines with dual plugs, etc.

rusefillc, 2021-07-09:
@mck1117 I've pushed two changes
1) TS project - outputs and constant angle setting as a temp solution
2) some of the output wiring no idea how much I've missed

angle setting is totally not used yet, not planning any more changes here in the next 24 hours

rusefillc, 2021-07-22:
```
/**
 * set engine_type 107
 */
void setRotary(DECLARE_CONFIG_PARAMETER_SIGNATURE) {
```

```
2021-07-21_20_38_16_721: EngineState: pin PC9: Trail 1
2021-07-21_20_38_16_721: EngineState: pin PC10: Primary Channel TX
2021-07-21_20_38_16_721: EngineState: pin PC11: Primary Channel RX
2021-07-21_20_38_16_722: EngineState: pin PD12: led: running status
2021-07-21_20_38_16_723: EngineState: pin PD13: led: warning status
2021-07-21_20_38_16_723: EngineState: pin PD14: led: CRITICAL status
2021-07-21_20_38_16_724: EngineState: pin PD15: led: comm status
2021-07-21_20_38_16_726: EngineState: pin PE10: Trail 2

```

commit 8b5dde5a (2021-07-22): trailing spark outputs fix #2931
commit c481f161 (2021-07-22): trailing spark outputs fix #2931
closed 2021-07-22 by rusefillc