rusefi/rusefi #2935: turbospeed input

rusefillc, 2021-07-09:
120krpm
 12compressor blades, sensor divide by factor of 8

if you theoretically have 8blade compressor you will have 1pulse per revolution 120k pulses per minute ie 2000hz

rusefillc, 2021-07-09:
andrey  2 hours ago
120krpm max, what is min?
New

Simon - Orchardperformance  21 minutes ago
Pretty much zero

Simon - Orchardperformance  21 minutes ago
Idling speed of the turbo can be sub 10k

Martin Matousek  16 minutes ago
aalso jacques sensor output is low (no pulses) below 5krpm or something, it probably allso have internal pullup to 5v (edited) 

Martin Matousek  15 minutes ago
and yes during idle turbospeed sensor can show 0krpm for sure and for smaller turbos 190 maybe even 200,210,220krpm is possible (edited) 

Simon - Orchardperformance  < 1 minute ago
Ours show a flicker from 1.5k to 2k.
Its pretty much flatline/ signal noise

Simon - Orchardperformance  < 1 minute ago
250k is right up at the top end of anything small available right now

rusefillc, 2021-07-27:
Related or not to https://github.com/rusefi/hellen76nissan-issues/issues/3?

rusefillc, 2021-07-27:
https://www.full-race.com/docs/Speed-Sensor-Installation-Instructions.pdf

mck1117, 2021-07-27:
250k rpm * 14 blades / 8 = 7.3khz

https://www.wolframalpha.com/input/?i=250000%2Fmin+*+14+%2F+8

mck1117, 2021-07-27:
an additional divide-by-8 would give us sub-1khz worst case

mck1117, 2021-07-27:
divide by 2 is easy with a D flip flop - chain more to divide by 2^n:

[image]

rusefillc, 2021-07-30:
Configuration fields added, next step is deciding if we copy/paste vehicle_speed.cpp or reuse vehicle_speed or else

See also - kind of depends on #3081