rusefi/rusefi #2942: pinouts process is broken but also green

rusefillc, 2021-07-10:
quick improvement: 

make it fail at least

```
processing ./firmware/config/boards/proteus/connectors/black35.yaml
DIR pinouts/proteus/
NAME black35
sed: -e expression #1, char 29: unknown command: `"'
pinouts/proteus//index.html: very short file (no magic)
IMG black35.jpg
black35.jpg
```

rusefillc, 2021-07-10:
@chuckwagoncomputing 

that's 2nd time yq changes on us. that means they will change on us again

can we dump specific yq source code or binary into https://github.com/rusefi/rusefi_external_utils/ so that we know for sure what version of yq we use at any moment?

chuckwagoncomputing, 2021-07-10:
I'd much rather have the workflow fetch a specific release on GH.

rusefillc, 2021-07-10:
works for me, as long as it's a bit more repeatable.

rusefillc, 2021-07-10:
@OrchardPerformance @BeerMoneyMotorsports we now have a manual one time backup at https://github.com/rusefi/rusefi_documentation/tree/master/pinouts-backup

OrchardPerformance, 2021-07-10:
I keep local, off line copies of some things incase of outage as well

chuckwagoncomputing, 2021-07-11:
Regarding keeping freezing software versions: the OS images that GH runs on update on a regular basis, with new versions of software. As far as I know, the only way to avoid this would be to self-host.

rusefillc, 2021-07-11:
but yq is not part of image is it?

we control where and how we get yq and yq specifically is the huge source of drama, not sed not java

rusefillc, 2021-07-13:
@chuckwagoncomputing close or do we have more plans here?

chuckwagoncomputing, 2021-07-13:
As far as I know we're done with this

closed 2021-07-13 by chuckwagoncomputing