rusefi/rusefi #2944: icon paths are broken when invoking via rusefi_autoupdate

rusefillc, 2021-07-10:
expected behavior:

``java -jar rusefi_console.jar ``
[image]

broken behavior, should be same as above
``java -jar rusefi_autoupdate.jar``
[image]

Come1LLF00, 2021-07-23:
Okay, I came up with the idea to add every URL of loaded jars including rusefi_console.jar to the system classloader when getClassLoaderByJar is invoked in AutoupdateUtil. The current solution looks like this:
`private static void addURLToSystemClassLoader( URL url ) throws IntrospectionException {
    URLClassLoader systemClassLoader = (URLClassLoader) ClassLoader.getSystemClassLoader();
    Class<URLClassLoader> classLoaderClass = URLClassLoader.class;

    try {
        Method method = classLoaderClass.getDeclaredMethod("addURL", URL.class );
        method.setAccessible(true);
        method.invoke(systemClassLoader, url );
    } catch ( Throwable t ) {
        t.printStackTrace();
        throw new IntrospectionException("Error when adding url to system ClassLoader ");
    }
}

@NotNull
public static URLClassLoader getClassLoaderByJar(String jar) throws MalformedURLException, IntrospectionException {
    final URL newURL = new File(jar).toURI().toURL();
    addURLToSystemClassLoader( newURL );
    return new URLClassLoader(
            new URL[]{ newURL },
            AutoupdateUtil.class.getClassLoader()
    );
}`

rusefillc, 2021-07-24:
@Come1LLF00 yes this looks fixed now!

closed 2021-07-24 by rusefillc