rusefi/rusefi #2945: WBO is setting itself to CAN

CrazyStriker, 2021-07-10:
CAN WBO seem to set itself to true.
Pushed the settings from TS a few times, resseting ECU in between and it's stuck to CAN input, hence no WBO input for me anymore.09/07/21 
Nightly build on H72

rusefillc, 2021-07-10:
that's pretty odd

Q1: just to confirm does your firmware match 100% match your project version?

Q2: just to reduce chance of misunderstanding please upload your very fresh tune?

Q3: just to reduce chance of misunderstanding please provide a URL for the exact dialog on the tune where you are trying to change the setting?

CrazyStriker, 2021-07-10:
- Project .ini is 2021/07/09
- https://rusefi.com/online/view.php?msq=579
- https://rusefi.com/online/view.php?msq=579&dialog=egoSettings

rusefillc, 2021-07-10:
https://github.com/rusefi/rusefi/issues/2944 is not a great example but can we please spell out 6 steps to reproduce, and spell out expected vs actual result?

CrazyStriker, 2021-07-10:
Issue timeline for me :
- Car was running on 2021/04/26 FW
- Downloaded nightly build on 07/10
- Updated .ini to 2021/07/09
- Set the board to DFU mode using TS
- Flashed the FW using .hex from 2021/07/09 nightly
- Reset the board

- Started TS on the right project. Showed a few errors regarding idle advance. Solved issue and reset board
- TS reads board configuration and generates difference report. 
Difference report shows this in local tune : 
[image]
And this in board tune : 
[image]

- I choose to push local tune tu set my WBO to analog
- Reset the board, and connected to TS. Goes back to difference report with same exact difference.
- While car is running, lambda shows 0

CrazyStriker, 2021-07-11:
[image]

PR #2946 (2021-07-11): fix hellen WBO override
fix #2945
closed 2021-07-11 by rusefillc