rusefi/rusefi #2948: dash pic example

duhafnusa4, 2021-07-11:
[image]
Thank you for taking the time to report an issue!

To make our work easier, please provide as many details as you can:
* [ ] What hardware did you use while experiencing the issue?

rusefillc, 2021-07-13:
@duhafnusa4 I am a bit lost here

what is the desire here on this ticket?

duhafnusa4, 2021-07-13:
This was made to host the pic so I could use it in the dash wiki

rusefillc, 2021-07-13:
@duhafnusa4 please close this ticket

@duhafnusa4 please consider discovering a less weird way to upload images :)

duhafnusa4, 2021-07-13:
Please provide me with non weird way and I will gladly :)

rusefillc, 2021-07-14:
This was discussed in slack

closed 2021-07-14 by rusefillc